BrunoLevy/learn-fpga #2: about the toolchain

wuxx, 2020-11-12:
Hi Bruno, I try to build these demos, but buid failed, I notice the gcc is riscv64-linux-gnu-gcc-10, which toolchain do you use?
I replace the toolchain with https://static.dev.sifive.com/dev-tools/riscv64-unknown-elf-gcc-20171231-x86_64-linux-centos6.tar.gz
and it report failed with this error
sysdeps/riscv/sfp-machine.h:21:10: fatal error: fpu_control.h: No such file or directory

BrunoLevy, 2020-11-12:
Hi,

I'm using two toolchains: the ubuntu package gcc-riscv64-linux-gnu and https://github.com/riscv/riscv-gnu-toolchain

For the runtime, it is a minimalist libc bundled in FIRMWARE/lib, that I assembled from pieces I copied here and there. It is not very clean, what you experience is probably clashes between system includes and mine. I did that because the one included in the ubuntu package does not have the integer multiplication functions (and hardware support for that in femtorv is optional, for instance if you use an IceStick).

I'll double check and come back to you. The plan is to have something cleaner (either with the standard gnu libc, or with a smaller one, several ones exist, need to test them).

BrunoLevy, 2020-11-12:
I have added the missing file in FIRMWARE/LIB/fpu_control.h, would you test again and tell me what it gives ?

wuxx, 2020-11-12:
sure, I test again, now it report the features.h missing
./fpu_control.h:22:10: fatal error: features.h: No such file or directory
 #include <features.h>

BrunoLevy, 2020-11-12:
Weird, in fact it is exactly the contrary of what I said (these files seem to be missing in your local installation of gcc, or more precisely, my stubs are not compatible with what your local installation of gcc expect)
Woud you try again ? I have pushed a quick and (very) dirty fix (replaced fpu_control.h with an empty file)

wuxx, 2020-11-12:
no it report endian.h missing, cound you try the toolchain I use above, it release by sifive so I think it's relatively universal
soft-fp/soft-fp.h:48:12: fatal error: endian.h: No such file or directory
 #  include <endian.h>

BrunoLevy, 2020-11-12:
OK, thank you, yes I'll test your toolchain and come back to you (your toolchain probably uses a different libc with files at different plaecs, I'll see what can be done)

BrunoLevy, 2020-11-12:
Just tried with the same toolchain as you and included another quick and dirty fix (dummy endian.h file)
Edit FIRMWARE/config.sh according to which toolchain you use and where it is installed.
I did not test the generated executables yet.

BrunoLevy, 2020-11-12:
Just tested my HelloWorld program with FemtoRV32 running on the IceStick. Everything seems to be allright. I would not bet that floating point numbers work though (will test them later on the ULX3S).

wuxx, 2020-11-13:
ok, later I will have a try

BrunoLevy, 2020-11-13:
Pushed another version: 
   - removed all floating-point routines from LIB
   - now linking with toolchain's libgcc.a to get the floating point routines
     (I could remove integer mutiplication and division routines as well, but I let them here for now, because 
      it can be nice to have a self-contained runtime for the IceStick)
 
See comments in FIRMWARE/config.sh

Tested with sifive and debian toolchains
Tested floating point routines

wuxx, 2020-11-29:
Hi, BrunoLevy, I try to build again, and it report the makefile.inc was missing.  in FIRMWARE/LIBFEMTORV32/Makefile
----------------
pi@raspberrypi:~/oss/learn-fpga/FemtoRV $ ./makeit_ulx3s.sh
======== Firmware
Missing FIRMWARE/firmware.hex
Using FemtOS image (needs OLED screen) (FIRMWARE/EXAMPLES/commander.exe)
To replace, cd FIRMWARE; ./make_firmware.sh EXAMPLES/....
============> Compiling libs
Makefile:26: ../makefile.inc: No such file or directory
make: *** No rule to make target '../makefile.inc'.  Stop.
Makefile:21: ../makefile.inc: No such file or directory
make: *** No rule to make target '../makefile.inc'.  Stop.
============> Making commander.exe
make: *** No rule to make target 'clean'.  Stop.
Something went wrong, change VERBOSE in make_firmware.sh and retry
======== Yosys
femtosoc.v:0: ERROR: Can not open file `FIRMWARE/firmware.hex` for \$readmemh.
pi@raspberrypi:~/oss/learn-fpga/FemtoRV $ cd FIRMWARE/
pi@raspberrypi:~/oss/learn-fpga/FemtoRV/FIRMWARE $ ./make_firmware.sh EXAMPLES/
commander.c              imgui_emul.h             sieve.c                  test_OLED.c
cube_OLED.c              imgui_road.c             spirograph_OLED.c        test_spi_flash.c
DATA/                    imgui_tunnel.c           ST_NICCC.c               test_spi_sdcard.c
demo_OLED.c              life_led_matrix.c        ST_NICCC_spi_flash.c     tinyraytracer.c
hello.c                  mandelbrot_float_OLED.c  sysconfig.c              TOOLS/
imgui_cup.c              mandelbrot_OLED.c        test_buttons.c           tty_OLED.c
imgui_doom.c             riscv_logo_OLED.c        test_font_OLED.c
pi@raspberrypi:~/oss/learn-fpga/FemtoRV/FIRMWARE $ ./make_firmware.sh EXAMPLES/hello.c
============> Compiling libs
Makefile:26: ../makefile.inc: No such file or directory
make: *** No rule to make target '../makefile.inc'.  Stop.
Makefile:21: ../makefile.inc: No such file or directory
make: *** No rule to make target '../makefile.inc'.  Stop.
============> Making hello
make: *** No rule to make target 'clean'.  Stop.
Something went wrong, change VERBOSE in make_firmware.sh and retry

BrunoLevy, 2020-11-29:
Thank you for notifying !
Silly me ! I had forgotten to push a couple of files, 
Now it is fixed (if I have not forgotten something else).

wuxx, 2020-11-29:
OK, keep focused, really interested with a risc-v below 1280 LUT 👍

BrunoLevy, 2020-12-08:
Just to male sure before closing the issue, does everything works now ?

wuxx, 2020-12-09:
ok, the compile is ok now

closed 2020-12-09 by wuxx