BrunoLevy/learn-fpga #29: Faster SPI flash controller

BrunoLevy, 2021-01-16:
There is (a lot of) room for improvement in the SPI flash controller:
- use a DDR block instead of freq/2 clock
- use faster SPI modes
- use dual IO SPI mode
- check whether the pins for quadspi are wired on the IceStick

BrunoLevy, 2021-01-18:
- Now using "fast read dual input" command.
- Tried to use "fast read dual IO" command (faster, saves 12 cycles by inputting address on two pins), no success for now
- Would be good to add XIP support (will save 8 additional cycles), but requires command to modify a register on the chip (will need additional logic, and can't afford the LUTs on the IceStick)

BrunoLevy, 2021-01-19:
- "fast read dual IO" command works
- Now working on XIP (adding a "control register", will see what it gives in terms of LUTs)

Mecrisp, 2021-01-25:
Trying the latest changes gives:

- touch firmware.hex
- Copied icestick_spi_flash_config.v to icestick_config.v
- make ICESTICK.synth

```
yosys -DICE_STICK -q -p "synth_ice40 -relut -top femtosoc -json femtosoc.json" RTL/femtosoc.v
Warning: Yosys has only limited support for tri-state logic at the moment. (RTL/DEVICES/MappedSPIFlash.v:50)
ERROR: Module `\EHXPLLL' referenced in module `\HDMI_clock' in cell `\pll_i' is not part of the design.
```

BrunoLevy, 2021-01-25:
Hi Matthias,
This is because the currently pushed version is configured for the ULX3S
To configure it for the IceStick:
Edit RTL/femtosoc_config.v and select CONFIGS/icestick_config.v instead of CONFIGS/ulx3s_config.v

BrunoLevy, 2021-01-25:
I meant CONFIGS/icestick_spi_flash_config.v
(and edit it as well to select the devices you want or not)

Mecrisp, 2021-01-25:
I see, thank you: Up and running now, feels faster !

BrunoLevy, 2021-01-25:
Cool ! Thank you for letting me know (I tested again, but I'm always afraid of breaking smthg ! When you have multiple board, it is very easy to break something without noticing !).

The current version uses 44 clocks for each 32-bits access, we could go faster:

- There is an "XIP" mode that uses 36 clocks and a configurable dummy cycles that could save another 4 cycles (32 cycles in the end), but I did not manage to activate them... 

- I could also probably need only 16 cycles for consecutive addresses (at the expense of a 22 bits reg + compare, which I probably cannot afford in terms of LUT budget).

Mecrisp, 2021-01-25:
You have a lot of configuration options, I like that ! I usually have a folder for each target separately, and common folder(s) for the parts that can be reused. Maintaining all the switches for different targets in the same target-specific files is something I would avoid. You'll end up with some duplicated glue code, but it improves readability for the ones that use one target only.

Speed is acceptable now, and you should leave a few LUTs for experiments of your students.

Did you follow the idea to change the processors register set in order to get 7 kb available RAM on Icestick ?

Mecrisp, 2021-01-25:
A good place to draw the line is between CPU+RAM in a common module, with options for size/SPI execution/reset vector, and a separate target-specific Verilog file to wire in the peripherals for each target.

In my experience with Mecrisp-Ice, I usually drop the same CPU/Memory/Forth module into my new projects and rewrite the IO part every time to fit the current use case, with the Forth core itself having a dependency on the terminal IO registers only.

BrunoLevy, 2021-01-25:
>Did you follow the idea to change the processors register set in order to get 7 kb available RAM on Icestick ?
Not yet, I mainly worked on the ULX3S version and its OS in the last few days. 

About files organization, I'm trying different things, no organization is perfect, I'll probably reorganize everything several times. For now it works reasonably well for the different projects, but I'll probably change it when the number of supported boards will increase. 

About LUTs, I have an idea that may save 40 to 50 LUTs: the SPI flash controller has a 40 bits and a 32 bits shift register, maybe I can replace them with 8-bits shift registers used several times (but I do not like it very much, it adds some complexity...)

BrunoLevy, 2021-01-26:
Pushed new version, now using a single 40 bits shifter (for both send and receive), and a single 5 bits clock counter, saves 50-80 LUTs !