BrunoLevy/learn-fpga #33: Difficulties to follow the IceStick Tutorial.

at91rm9200, 2021-02-21:
Hello Bruno,

according to the IceStick Tutorial, I executed “make ICESTICK”. This leads to the following error message on my system (Ubuntu 18.04):

/home/bernd/fpga/mue/project/riscv/femtorv_2102/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/riscv64-unknown-elf-gcc-8.3.0-2020.04.0-x86_64-linux-ubuntu14/bin/riscv64-unknown-elf-as -march=rv32i -mabi=ilp32 -defsym SSD1351=1 -defsym ICE_STICK=1 -I/home/bernd/fpga/mue/project/riscv/femtorv_2102/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/bernd/fpga/mue/project/riscv/femtorv_2102/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/bernd/fpga/mue/project/riscv/femtorv_2102/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC  ssd1351_1331.S -o ssd1351_1331.o 
make[2]: *** No rule to make target 'ssd1351_1331_init.o', needed by 'libfemtorv32.a'.  Stop.

Since there is no  ssd1351_1331_init.c, I removed the dependency in  LIBFEMTORV32/Makefile
and the libs were compiled.

If I run “make ICESTICK” again, I get the next error message, which indicates, that the firmware.hex is missing: 

==== Generated femtorv32 libs.
make[1]: Leaving directory '/home/bernd/fpga/mue/project/riscv/femtorv_2102/learn-fpga/FemtoRV/FIRMWARE'
yosys -DICE_STICK -q -p "synth_ice40 -relut -top femtosoc -json femtosoc.json" RTL/femtosoc.v
Warning: Yosys has only limited support for tri-state logic at the moment. (RTL/DEVICES/MappedSPIFlash.v:268)
RTL/femtosoc.v:0: ERROR: Can not open file `FIRMWARE/firmware.hex` for \$readmemh.
BOARDS/icestick.mk:11: recipe for target 'ICESTICK.synth' failed
make: *** [ICESTICK.synth] Error 1

Could you please look into this?

Regards, Bernd.

BrunoLevy, 2021-02-21:
It's because I am stupid !
I forgot to push this file,
Now normally it is fixed,
Would you try again and tell me what it gives ?
Thanks in advance,
-- B

at91rm9200, 2021-02-21:
Thank you very much. I works perfectly now.

Regards, Bernd.

at91rm9200, 2021-02-21:
Hello Bruno,

sorry to bother you again. I am not able to see any UART output in the terminal.

Executing make ICESTICK in FemtoRV directory works without errors,  iceprog is successfully programming femtosoc.bin. But there is no output in the terminal.

make blinker_loop.prog in the ASM_EXAMPLES directory works as expected. I can see the LEDs blink.

Make hello.prog in the EXAMPLES directory seems not to produce any output in the terminal.

So I disabled/commented NRV_IO_SSD1351 and NRV_IO_MAX7219 in RTL/CONFIGS/icestick_config.v and run make ICESTICK again.

Now I get the following error message:

/home/bernd/fpga/mue/project/riscv/femtorv_2102/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/riscv64-unknown-elf-gcc-8.3.0-2020.04.0-x86_64-linux-ubuntu14/bin/riscv64-unknown-elf-gcc -Os -I/home/bernd/fpga/mue/project/riscv/femtorv_2102/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/bernd/fpga/mue/project/riscv/femtorv_2102/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/bernd/fpga/mue/project/riscv/femtorv_2102/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax  -c femtoGL.c
femtoGL.c: In function 'GL_init':
femtoGL.c:62:16: error: 'OLED_WIDTH' undeclared (first use in this function)
    GL_width  = OLED_WIDTH;
                ^~~~~~~~~~
femtoGL.c:62:16: note: each undeclared identifier is reported only once for each function it appears in
femtoGL.c:63:16: error: 'OLED_HEIGHT' undeclared (first use in this function)
    GL_height = OLED_HEIGHT;
                ^~~~~~~~~~~
../makefile.inc:72: recipe for target 'femtoGL.o' failed
make[2]: *** [femtoGL.o] Error 1

I am unsure now, if  NRV_IO_SSD1351 and NRV_IO_MAX7219 should be disabled or not.

Regards, Bernd.

at91rm9200, 2021-02-21:
I found a workaround: I added

#if !defined(SSD1351) || !defined(SSD1331)
#define OLED_WIDTH  1
#define OLED_HEIGHT 1
#endif  

in femtorv32h, line 112 to get femtoGL compiled with NRV_IO_SSD1351 and NRV_IO_MAX7219 disabled. 
And now I can see the UART messages in the terminal :-)

Regards, Bernd.

BrunoLevy, 2021-02-22:
Hi,
There is an easier way:
You can edit RTL/CONFIGS/icestick_config.v and comment-out all the devices you are not using 
(NRV_IO_SSD1351, NRV_IO_MAX7219), then the firware detects it and sends all outputs to the UART.

I'll make it clearer in the tutorial.

Best wishes,
-- B

at91rm9200, 2021-02-22:
yes, I tried to comment-out NRV_IO_SSD1351 and NRV_IO_MAX7219. But compiling the firmware fails then with the error, I mentioned above: 
femtoGL.c: In function 'GL_init':
femtoGL.c:62:16: error: 'OLED_WIDTH' undeclared (first use in this function)
GL_width = OLED_WIDTH;

Anyway, I thank you for this fantastic project and close this issue now.
Regards, Bernd.

closed 2021-02-22 by at91rm9200

BrunoLevy, 2021-02-22:
You are perfectly right.
 I have pushed your suggested fix, thank you very much !
-- B