BrunoLevy/learn-fpga #39: ICESUGAR target fails to build

piso77, 2021-05-03:
learn-fpga/FemtoRV$ make ICESUGAR
...
make[2]: Leaving directory '/home/flag/fpga_stuff/learn-fpga/FemtoRV/FIRMWARE/CRT'
==== Generated femtorv32 libs.
make[1]: Leaving directory '/home/flag/fpga_stuff/learn-fpga/FemtoRV/FIRMWARE'
make: *** No rule to make target 'FIRMWARE/firmware.hex', needed by 'ICESUGAR.synth'.  Stop.

BrunoLevy, 2021-07-11:
Yes you'll need to adapt icesugar.mk and create RTL/CONFIGS/icesugar_config.v plus a couple of other changes (I could not test it with the icesugar). It is not very difficult to do, let me know if you want to do it / test it and I'll help you.

abderraouf-adjal, 2022-04-12:
@piso77 Hi, where you able to do a core on icesugar or icesugar-nano on the SPI flash?

piso77, 2022-04-13:
I have a series of commits to enable ICESUGAR, but it never really worked - not sure if if was the flash chip or something in my memory map. Should give it another shot, or prepare a pull request if you want to try it out.

abderraouf-adjal, 2022-04-13:
I had a luck running a laggy UART hello word from old version using the
firmware.hex.

New versions of this repo is broken for icesugar, managed to edit for
successful compile, but no luck to see it alive.

abderraouf-adjal, 2022-04-29:
@piso77 checkout #66 for icesugar-nano, it works fine in my end, I don't have an icesugar board to test on it, however it's not very different, just re-naming and make sure the SPI pins in .PCF file are correct by looking at the board schematic.

piso77, 2022-04-30:
> @piso77 checkout #66 for icesugar-nano, it works fine in my end, I don't have an icesugar board to test on it, however it's not very different, just re-naming and make sure the SPI pins in .PCF file are correct by looking at the board schematic.

Thanks,  i'll give it a shot later today or tomorrow.