BrunoLevy/learn-fpga #41: typo in the verilog file

KiranKanchi, 2021-06-27:
Dear Bruno,

RTL/DEVICES/MappedSPIFlash.v:306: ERROR: syntax error, unexpected ';'

Kind regards
KiranKanchi

KiranKanchi, 2021-06-27:
I also see the follow error when i do 'make ICESTICK': 

==== Generated femtorv32 libs.
make[1]: Leaving directory '/home/kirankanchi/learn-fpga/FemtoRV/FIRMWARE'
yosys -DICE_STICK -q -p "synth_ice40 -relut -top femtosoc -json femtosoc.json" RTL/femtosoc.v
Warning: Yosys has only limited support for tri-state logic at the moment. (RTL/DEVICES/MappedSPIFlash.v:287)
nextpnr-ice40 --force --json femtosoc.json --pcf BOARDS/icestick.pcf --asc femtosoc.asc --freq 12 --hx1k --package tq144 --opt-timing
icetime -p BOARDS/icestick.pcf -P tq144 -r femtosoc.timings -d hx1k -t femtosoc.asc 
// Reading input .pcf file..
Can't open input file: No such file or directory

Kind regards
KiranKanchi

BrunoLevy, 2021-07-11:
You are perfectly right ! (yosys does not complain, but iverilog indicates a syntax error). Fixed.

closed 2021-07-11 by BrunoLevy