BrunoLevy/learn-fpga #50: Clarification on SDRAM chip on ULX3S for Litex

darkstar007, 2021-12-30:
Hi,
I was following  `learn-fpga/FemtoRV/TUTORIALS/litex.md` and it says: 
```
determine SDRAM chip (see image): one of MT48LC16M16, AS4C32M16 or AS4C16M16
```
However, mine, like the majority of ULX3S has a  IS42S16160G in them (see [here](https://github.com/emard/ulx3s/blob/master/doc/MANUAL.md) at the bottom of the page)

Could these instructions be updated to state what value the SDRAM variable should be set to for this case?

Thanks,
Matt

BrunoLevy, 2021-12-30:
Hi Matt,

From what I'm seeing in the tech sheet, it seems that  IS42S16160G is very similar to the other chips supported by LiteX (except that it supports higher freq). There are good chances that it can work with one of the other chips supported by LiteX, did you try ? LiteX BIOS has a sdram_test command that can help detecting if something goes wrong.

Best,
-- Bruno

darkstar007, 2021-12-30:
Hi Bruno,

sdram_test didn't work with AS4C32M16, it worked with MT48LC16M16 (but only tested first megabyte - not sure if that's expected). Not tried AS4C16M16 yet - got sidetracked by food....will try shortly.

Thanks,
Matt

darkstar007, 2021-12-30:
AS4C16M16 also appears to work - and also only 1Mbyte is tested.

BrunoLevy, 2021-12-30:
LiteX BIOS has several commands to do various tests. 

litex> mem_list
will show the different memories projected into address space.
MAIN_RAM is the SDRAM (starts at address 0x40000000)

then you can do:
mem_test <addr> <size>

you can also do:
mem_speed <addr> <size>

darkstar007, 2021-12-31:
Using MT48LC16M16 I get 
```
litex> mem_list
Available memory regions:
ROM       0x00000000 0x20000 
SRAM      0x01000000 0x2000 
MAIN_RAM  0x40000000 0x2000000 
CSR       0x82000000 0x10000 

litex> mem_test 0x40000000 0x2000000 
Memtest at 0x40000000 (32.0MiB)...
  Write: 0x40000000-0x42000000 32.0MiB    
   Read: 0x40000000-0x42000000 32.0MiB    
Memtest OK

litex> mem_speed 0x40000000 0x2000000 
Memspeed at 0x40000000 (Sequential, 32.0MiB)...
  Write speed: 4.8MiB/s
   Read speed: 6.4MiB/s

```
So its all probably working....

BrunoLevy, 2021-12-31:
Sounds good !
Thank you for the update.
Normally these memory chips share many characteristic. You'll tell me whether the demos work (added the missing demos.h file), and I'll add a note in the doc that 
"--sdram-module MT48LC16M16" can work for  IS42S16160G.

darkstar007, 2021-12-31:
Just tried the demos and they seem fine.

closed 2022-01-01 by BrunoLevy