BrunoLevy/learn-fpga #51: Demos do not compile

darkstar007, 2021-12-31:
Hi,

It looks like there is a missing include file:
```
$ cd learn-fpga/FemtoRV/FIRMWARE/LiteX/DemoBundle
$ make BUILD_DIR=/home/matt/litex_2022/build/radiona_ulx3s/
 CC       tinyraytracer.o
demos/tinyraytracer.c:6:10: fatal error: demos.h: No such file or directory
 #include "demos.h"
          ^~~~~~~~~
compilation terminated.
make: *** [Makefile:110: tinyraytracer.o] Error 1
```
I cannot see demos.h in the repos.....

Thanks, 
Matt

BrunoLevy, 2021-12-31:
Hi Matt,
Oops, Forgot to add this file last time. Now I've added it.
Best regards,
-- B

darkstar007, 2021-12-31:
Hi B,

Thanks, but I now get:
```
$ make BUILD_DIR=/home/matt/litex_2022/build/radiona_ulx3s/
 CC       tinyraytracer.o
In file included from demos/tinyraytracer.c:6:
demos/demos.h:4:10: fatal error: command.h: No such file or directory
 #include "command.h"
          ^~~~~~~~~~~
compilation terminated.
make: *** [Makefile:110: tinyraytracer.o] Error 1
```

Matt

BrunoLevy, 2021-12-31:
On the command line you need to specify where LiteX is installed:
```
$ make LITEX_DIR=<your local installation of LiteX> LITEX_PLATFORM=radiona_ulx3s (or whatever you are using)
```

Alternatively, you can edit the Makefile and make it point to your local installation of LiteX:
```
# find BUILD_DIR without using Python wrapper
LITEX_DIR?=/home/blevy/Programming/LiteX
LITEX_PLATFORM?=radiona_ulx3s
```

darkstar007, 2021-12-31:
Ahh, OK, I was taking the instructions from [here](https://github.com/BrunoLevy/learn-fpga/tree/master/FemtoRV/FIRMWARE/LiteX) where it says:
```
$ make BUILD_DIR=<path where you synthesized LiteX> 
```

It builds now.

BrunoLevy, 2021-12-31:
Thanks ! (I've fixed the documentation that was wrong !)

darkstar007, 2021-12-31:
I also came across some typo's you might want to fix. On [this](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/TUTORIALS/toolchain.md), in the section on `OpenFPGALoader` every command has `OpenFPGALoader` spelt as `openFLGALoader`!

darkstar007, 2021-12-31:
Sorry, another typo in [this](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/FIRMWARE/LiteX/README.md) page. In Step 2: Execute, it says:
```
$ lxterminal --kernel demo.bin /dev/ttyUSBnn
```
and should be:
```
$ lxterm  --kernel demo.bin /dev/ttyUSBnn
```

BrunoLevy, 2021-12-31:
Many thanks for your feedback and for the fixes !

closed 2022-01-01 by BrunoLevy