BrunoLevy/learn-fpga #55: How to add my own peripheral?

ladmanj, 2022-01-26:
Hi to all
Thanks for this compact package, I greatly appreciate that i got everything running on my icebreaker board with basically two to three commands.
I am now searching for information how to add my own peripheral.
I need to instantiate two instances of my module. Every single module needs two pins for input and output via PMOD interface talking to the external world, one clock source (main CPU clock will be fine) and it creates single 32-bit register to be read from the CPU.

I still didn't found how the IO pins are allocated, i have them in .pcf file with some names, but I don't know how to direct it to the module.

The implementation of the module was tested with simple testbench acting as top verilog module with every IO connected to pins.

Thank you for a bit of help.

BrunoLevy, 2022-01-26:
To use a IO pin, you need first to know what is the ID of the pin. On the IceBreaker, pin IDs are numbers. You can find them in the following picture (FPGA pin numbers):
https://github.com/icebreaker-fpga/icebreaker/raw/master/img/icebreaker-v1_0b-legend.jpg

Then you can edit BOARDS/icebreaker.pcf and add your pins, giving them names (names will be the way to refer to the pin in the Verilog file). 
For instance, to use the four rightmost pins of PMOD1B 
set_io module1_in 31
set_io module1_out 28
set_io module2_in 34
set_io module2_out 32

Then you will need to add module1_in,module1_out,module2_in,module2_out to the interface of femtosoc (in RTL/femtosoc.v):
module femtosoc(
   ...
   input module1_in,
   output module1_out,
   input module2_in,
   output module2_out
)

Then you have your register that needs to be mapped to the address space of the CPU, so that it can be read/written.
There is a big comment in femtosoc.v that explains how it works / how to do that (starting from line 320)

Hope this helps,
Best,
-- B

ladmanj, 2022-02-03:
Hi @BrunoLevy 

Thank you
My module is now integrated and it definitely does something.

Unfortunately it somehow interferes with uart, but no pins nor address bits are shared.

I'm guessing that there is probably overlap in spiflash as the hardware is getting bigger.
I can't find where the boundaries are defined.

Thank you for an advice
J.

ladmanj, 2022-02-03:
Hi @BrunoLevy 

Actually it is not overlapping 

````
iceprog femtosoc.bin
init..
cdone: high
reset..
cdone: low
flash ID: 0xEF 0x40 0x18 0x00
file size: 104090
erase 64kB sector at 0x000000..
erase 64kB sector at 0x010000..
````

````
iceprog -o 128k main.spiflash.bin
init..
cdone: high
reset..
cdone: low
flash ID: 0xEF 0x40 0x18 0x00
file size: 2392
erase 64kB sector at 0x020000..
````

In that case I don't know what's wrong :-(

Thank you

ladmanj, 2022-02-03:
Even with the clock speed decreased down to 12MHz (with PLL disabled), the uart transmission is damaged.

ladmanj, 2022-02-03:
That one problem with uart now seems to be fixed, the common IO data bus was compromised.

Mecrisp, 2022-02-09:
I found that someone made an intro exactly to answer your question:
https://www.youtube.com/watch?v=DtAwbKqLA5Y
https://github.com/ShawnHymel/introduction-to-fpga

Mecrisp, 2022-02-09:
Sorry, Bruno already mentioned that one before :-)

closed 2023-06-15 by BrunoLevy