BrunoLevy/learn-fpga #57: Disable auto remove the .o .elf .bin files

NaitikSanas, 2022-02-22:
After compiling and uploading the program using command `make main.prog` the output files of compiler is removed and We right now need them for debugging purpose. so how can we disable the auto remove function. I tried to comment out the line `168` of `makefile.inc` which seems to be removing but that didn't worked. any help will be appreciated.

NaitikSanas, 2022-02-22:
Hey, We managed to get the .o elf and bin files using command you suggested. Thank you for your help and support!

BrunoLevy, 2022-02-22:
If you want to automatically keep intermediate files, it is also possible:
edit `FIRMWARE/makefile.inc` and add the following line at the end of the file:
`.SECONDARY:`
(then it will mark all generated files as "secondary targets", which means `make` will keep them)
More details here:
https://stackoverflow.com/questions/30997638/makefile-rule-to-keep-intermediate-files

closed 2022-02-22 by BrunoLevy