BrunoLevy/learn-fpga #63: Not able to get any output from UART using Icestick

r1cebank, 2022-03-27:
Hi,

Thank you for building this awesome project, I am trying to build the softcore for Icestick and run an example program but I am not able to get any UART output. I verified the core built on the Icestick works by uploading a simple program that blinks the LED onboard, but for some unknown reason I can't get the USB serial to output anything.

I am building from commit: f681e81b1ea41b8d77649df8076121ec009910e3

Here is the config I am using, I've disabled everything except, LED, UART and SPI Flash

```
`define NRV_IO_LEDS          // Mapped IO, LEDs D1,D2,D3,D4 (D5 is used to display errors)
// `define NRV_IO_IRDA          // In IO_LEDS, support for the IRDA on the IceStick (WIP)
`define NRV_IO_UART          // Mapped IO, virtual UART (USB)
// `define NRV_IO_SSD1351       // Mapped IO, 128x128x64K OLED screen
// `define NRV_IO_MAX7219       // Mapped IO, 8x8 led matrix
`define NRV_MAPPED_SPI_FLASH // SPI flash mapped in address space. Can be used to run code from SPI flash.
```

Thank You

r1cebank, 2022-03-27:
Here is the source code for my test program, the LED blinks but no output in the serial console

```C
#include <femtorv32.h>

int main() {

   while(1) {
     printf("Hello world !!\n Let me introduce myself, I am FemtoRV32, one of the smallest RISC-V cores\n");
     printf("Freq: %d MHz\n", FEMTORV32_FREQ);
     delay(500);
     *(volatile uint32_t*)(0x400004) = 3;
     delay(500);
     *(volatile uint32_t*)(0x400004) = 0;
   }

   return 0;
}

```

BrunoLevy, 2022-03-28:
Hello, 
Your configuration and program looks correct to me,
Which commands did you run ?
For running the example:
```
  $ cd FemtoRV
  $ make IceStick  (this will synthesize and send the bitstream to the device)
  $ cd FIRMWARE/EXAMPLES
  $ make hello.prog (this will compile the program and send the binary to SPI flash)
  $ cd ../../
  $ make terminal (or use your favorite terminal emulator, at 115200 bauds)
```
Hope this helps,
Best,
-- Bruno

r1cebank, 2022-03-28:
Hello, thanks for the quick reply.

Yes, thats the commands i used to build the core and the program. The LED works but nothing is printed on the serial USB

BrunoLevy, 2022-03-28:
There could be several reasons:
- check which device is used (ttyUSB0, ttyUSB1, ....) by plugging the device then `$ dmesg`
- try different terminal emulators
- try https://github.com/BrunoLevy/learn-fpga/tree/master/FemtoRV/TUTORIALS/FROM_BLINKER_TO_RISCV
(step18: `$ BOARDS/run_icestick.sh step18.v` then `terminal.sh`)

r1cebank, 2022-03-28:
Hi, I've already checked the device for the icestick and made sure I am using the correct one `ttyUSB1`, the terminal emulators I've tried are `screen` `picocom`.

I tried running `step18.v` as instructed, and check the serial output, the LED blinks rapidly for a while and goes off but nothing is printed in the terminal.

Thanks

BrunoLevy, 2022-03-28:
I don't have any other idea. Just in case, I'd try with `ttyUSB0` and `ttyUSB2` (but you probably tryed already...)
Could be also a question of permissions ? Or udev rules ? (but obviously programming the device works, so probably not).

Mecrisp, 2022-03-28:
Maybe try to toggle DTR which is connected to reset line?

BrunoLevy, 2022-03-28:
`miniterm --dtr=0 ttyUSB0 115200` maybe ?

r1cebank, 2022-03-28:
Thanks for the sugesstions, I've tried `miniterm --dtr=0 ttyUSB0 11520` as well since its one of the option in the Makefile. I'll do some my own digging.

wohlbier, 2023-01-12:
Hi @r1cebank did you get this to work? For step 18 I'm seeing question marks written to the terminal. Is it supposed to be text?