BrunoLevy/learn-fpga #64: Compile problem

darkstar007, 2022-04-18:
Coming back to this after a couple of months and installing new versions of everything (Litex/yosys/nextptr) from github, and now it won't compile (yosys error):
```
~/learn-fpga/LiteX$ python3 -m boards.radiona_ulx3s_ex --device LFE5U-85F --with-oled --sdram-module MT48LC16M16 --cpu-type femtorv --cpu-variant petitbateau --build --load --ecppack-compress

[snip]
Info:           ECLKBRIDGECS:     0/    2     0%
Info:                   DCSC:     0/    2     0%
Info:             TRELLIS_FF:  3536/83640     4%
Info:           TRELLIS_COMB: 13702/83640    16%
Info:           TRELLIS_RAMW:   131/10455     1%

ERROR: Cell 'wifi_en$tr_io' cannot be bound to bel 'X0/Y47/PIOB' since it is already bound to cell 'wifi_gpio0$tr_io'
0 warnings, 1 error
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/matt/github_cloned/learn-fpga/LiteX/boards/radiona_ulx3s_ex.py", line 138, in <module>
    main()
  File "/home/matt/github_cloned/learn-fpga/LiteX/boards/radiona_ulx3s_ex.py", line 131, in main
    builder.build(**builder_kargs, run=args.build)
  File "/home/matt/litex_2022/litex/litex/soc/integration/builder.py", line 350, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/matt/litex_2022/litex/litex/soc/integration/soc.py", line 1198, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/matt/litex_2022/litex/litex/build/lattice/platform.py", line 43, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/matt/litex_2022/litex/litex/build/lattice/trellis.py", line 236, in build
    _run_script(script)
  File "/home/matt/litex_2022/litex/litex/build/lattice/trellis.py", line 174, in _run_script
    raise OSError("Error occured during Yosys/Nextpnr's script execution.")
OSError: Error occured during Yosys/Nextpnr's script execution.
```

Commenting out 
```
     soc.add_ESP32()
```
in radiona_ulx3s_ex.py made everything happy again....

BrunoLevy, 2022-04-18:
Thank you for this report, it is probably an evolution of LiteX that has now a wire to this pin, I'll double check.

BrunoLevy, 2023-06-14:
Deactivated dynamic ESP32 switch for now

closed 2023-06-14 by BrunoLevy