BrunoLevy/learn-fpga #73: fail in make ICESTICK

steven07210011, 2022-09-07:
when i run make ICESTICK and it show fail
./PROCESSOR/femtorv32_quark.v:68: warning: Attributes are not supported on net declaration assignments and will be discarded.
ARCH=rv32i
OPTIMIZE=-Os
ABI=ilp32
RAM_SIZE=       6144
DEVICES= -DICE_STICK=1
BOARD=icestick
(cd FIRMWARE; make libs)
make[1]: Entering directory '/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE'
(cd LIBFEMTOGL;   make clean all) # Compile graphic library
make[2]: Entering directory '/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE/LIBFEMTOGL'
Configuring for Raspberry Pi
../makefile.inc:98: warning: ignoring prerequisites on suffix rule definition
../makefile.inc:102: warning: ignoring prerequisites on suffix rule definition
../makefile.inc:106: warning: ignoring prerequisites on suffix rule definition
rm -f *.o *.elf *.hex *.exe *~ *.a *.bin *.list
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-as -march=rv32i -mabi=ilp32 -defsym ICE_STICK=1 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC  font_8x16.S -o font_8x16.o 
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-as -march=rv32i -mabi=ilp32 -defsym ICE_STICK=1 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC  font_8x8.S -o font_8x8.o 
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-as -march=rv32i -mabi=ilp32 -defsym ICE_STICK=1 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC  font_5x6.S -o font_5x6.o 
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-as -march=rv32i -mabi=ilp32 -defsym ICE_STICK=1 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC  font_3x5.S -o font_3x5.o 
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c femtoGL.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c femtoGLtext.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c femtoGLfill_rect.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c femtoGLsetpixel.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c femtoGLline.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c femtoGLfill_poly.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c tty_init.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c max7219_text.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c FGA_mode.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c FGA.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-gcc -Os -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC -DICE_STICK=1 -fno-pic -march=rv32i -mabi=ilp32 -fno-stack-protector -w -Wl,--no-relax   -c femto_GUI.c
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-ar cq libfemtoGL.a font_8x16.o font_8x8.o font_5x6.o font_3x5.o femtoGL.o femtoGLtext.o femtoGLfill_rect.o femtoGLsetpixel.o femtoGLline.o femtoGLfill_poly.o tty_init.o max7219_text.o FGA_mode.o FGA.o femto_GUI.o 
/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-8.3.0-2.3-linux-arm/bin/riscv-none-embed-ranlib libfemtoGL.a
make[2]: Leaving directory '/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE/LIBFEMTOGL'
(cd LIBFEMTORV32; make clean all) # Compile hardware support lib
make[2]: Entering directory '/home/pi/Projects/fpga/learn-fpga/FemtoRV/FIRMWARE/LIBFEMTORV32'
Configuring for Raspberry Pi
../makefile.inc:98: warning: ignoring prerequisites on suffix rule definition
../makefile.inc:102: warning: ignoring prerequisites on suffix rule definition
../makefile.inc:106: warning: ignoring prerequisites on suffix rule definition
rm -f *.o *.elf *.hex *.exe *~ *.a *.bin *.list
[... cut ...]
Info: Creating initial analytic placement for 1123 cells, random placement wirelen = 18007.
Info:     at initial placer iter 0, wirelen = 522
Info:     at initial placer iter 1, wirelen = 563
Info:     at initial placer iter 2, wirelen = 555
Info:     at initial placer iter 3, wirelen = 524
Info: Running main analytical placer.
ERROR: Failed to expand region (0, 0) |_> (13, 17) of 1297 ICESTORM_LCs
11 warnings, 1 error
make: *** [BOARDS/icestick.mk:12: ICESTICK.synth] Error 255

may i know how to solve it?

BrunoLevy, 2022-09-07:
Seems that depending on the version of yosys the same design will fit on the icestick or not. Try deactivating the devices that you do not need (OLED display, LED matrix if you do not use them). 
I'll investigate on my side and tell you...

steven07210011, 2022-09-07:
> Seems that depending on the version of yosys the same design will fit on the icestick or not. Try deactivating the devices that you do not need (OLED display, LED matrix if you do not use them). I'll investigate on my side and tell you...

Ok take noted