BrunoLevy/learn-fpga #74: Portability of tutorials to Upduino

hexad, 2022-10-14:
I want to learn FPGA development and these tutorials are amazing - thank you so much!

However, there are zero available IceSticks in the market right now. A cheap and accessible FPGA is the Upduino.
Is it possible that these tutorials can be transferrable to something like the Upduino?

Pardon my lack of knowledge - but this is the same reason I came across your repository.

Mecrisp, 2022-10-14:
Welcome here! As far as I know the Upduino is based on Lattice UP5K... While not having a solution for porting Brunos complete work, I wrote a RISC-V playground for the MCH2022 badge that uses the same FPGA and the FemtoRV32-Gracilis:

https://github.com/badgeteam/mch2022-firmware-ice40/tree/master/projects/RISCV-Playground
https://www.badge.team/docs/badges/mch2022/software-development/risc-v/

Maybe this gives you a starting point? There are also more simple examples if you explore the mch2022-firmware-ice40 repository a bit.

Best wishes for your journey into FPGAs that might soon become a deep rabbit hole for you to explore :-)

Mecrisp, 2022-10-14:
Wait, my error: "Icebreaker" is supported out of the box and much closer to your Upduino. But let Bruno comment on that!

BrunoLevy, 2022-10-14:
Yep, it is probably quite easy to adapt icebreaker support to Upduino
Internal clock is 12 MHz for both, cool

1)
```
  $ cd learn-fpga/FemtoRV/BOARDS
  $ cp icebreaker.mk upduino.mk
  $ cp icebreaker.pcf upduino.pcf
```
Edit `upduino.mk`, replace all instances of icebreaker with upduino (respect caps)

2) correct upduino.pcf with the correct pins fpr pclk, D1...D5, TXD,RDX and RESET

3) edit `RTL/femtosoc_config.v` and add a config for UPDUINO

4)  
```
   $ cd learn-fpga/FemtoRV/RTL/configs
   $ cp icebreaker_config.v upduino_config.v
```
See if the reset button is negative or positive (if negative, define NRV_NEGATIVE_RESET)
If nothing happens, then try maintaining RESET (if it works, it means you need to define NEGATIVE_RESET)

Normally it will be easy, the most tricky part will be finding the pin numbers for everything. The SPI Flash can be a bit tricky if you want to enable it in dual IO mode (but you can start with regular mode)

hexad, 2022-10-23:
Thank you very much for this repo and for the support!
However, i am a bit lost with the repo structure.
As far as i understand the `Basic` folder contains some random examples (listed at the end of the main README.md).
Then the main starting points are the two READMEs under the From `Blinky to RISC-V` category (Episode I and Episode II)?
Is development of femtoRV covered in those documents?
What are the `Attic`, `Basic` and `FemtoRV` folders? Is the `FemtoRV` folder the main starting point? It also contains a bunch of READMEs inside.
Thank you once again and sorry for the trouble!

BrunoLevy, 2022-10-29:
It is because there are different projects all in the same place (maybe I should create separate repositories)
- Step1: *Basic* it is a set of simple and self-contained projects to test a specific board, I'll start by adding a Upduino subdirectory here, and then create a Blinker 
- Step 2: *FemtoRV* there you have a complete SOC with support for different peripherals. Add the files for your board in the BOARDS subdirectory, and add targets in the Makefile. More details [here](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/TUTORIALS/newboard.md)
- Step 3: *From blinker to RISC-V* this tutorial is meant to become a course / book on the topic. It has step by step instructions to create a pipelined RISC-V processor. It is also possible to add support for additional boards.