BrunoLevy/learn-fpga #75: system miss the libboost_filesystem.so.1.71.0

ZhangPeterGree, 2022-11-13:
[image]
working on path: ~/litex-env/learn-fpga/FemtoRV$  make ICESTICK
...  ...
nextpnr-ice40 --force --json femtosoc.json --pcf BOARDS/icestick.pcf --asc femtosoc.asc --freq 12 --hx1k --package tq144 --opt-timing

nextpnr-ice40: error while loading shared libraries: libboost_filesystem.so.1.71.0: cannot open shared object file: No such file or directory

make: *** [BOARDS/icestick.mk:12: ICESTICK.synth] Error 127

I tried to install it, and it said the installation was successful. But I still can't find it, how do I set the path?
https://www.boost.org/users/history/version_1_71_0.html

BrunoLevy, 2022-11-13:
How did you install nextpnr-ice40 ?
Did you follow the protocol described [here](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/TUTORIALS/toolchain.md) ?
What does it say when you do:
```
  $ find / -name libboost_filesystem.so.1.71.0 -print
```
?

ZhangPeterGree, 2022-11-14:
Make sure it is not perfect nextpnr-ice40

closed 2022-11-14 by ZhangPeterGree