BrunoLevy/learn-fpga #87: Step 7 Yields No Output

usmank11, 2023-01-08:
Hi, Im going through the tutorial and everything was working just fine until I got to step 7. 

For context I am using a PYNQ Z2 development board in Vivado 2018.1

I am using exact copies of the necessary files, but the step 7 file doesn't show anything for the LED's in the simulation other than red lines with X's and doesn't light up any LED's. I tried further steps by directly copying their files in but I had no luck with getting any of them to work. Does anyone know what my issue could be?

BrunoLevy, 2023-01-08:
Hi, 

I am using the open-source tools for this tutorial (Yosys/Nextpnr/Verilator/Icarus), I haven't tested yet with Vivado.
Seeing 'X's in simulation means that some registers were not initialized, but it is surprising, because I do not encounter the problem with Icarus.

Would you send the error messages that you get with commented-out `default_nettype none` ? This may contain a hint.

usmank11, 2023-01-08:
There are no errors with it commented out and it simulates with the X's. If I uncomment it shows this same net type error for my modules input/output.
[image]

I commented out TXD as its not needed for now but that shouldn't be causing any issues.

BrunoLevy, 2023-01-08:
I think it wants `input wire CLK` instead of `input CLK` (but it is not the cause of the X's)

usmank11, 2023-01-08:
Yea that works if I add wire to the inputs/outputs of both the main file and clockworks in step 6, but still have the same issue as before in step 7.

BrunoLevy, 2023-01-08:
So it is probably something that I'm doing in `riscv_assembly.v` that works with Yosys, with Verilator and with Icarus but that does not work with Vivado.
I'll take a look and test it with Vivado as soon as I find some time. 
In the meanwhile, you may try step20: starting from step20, the tutorial uses the GNU toolchain instead of my builtin RISC-V assembler written in Verilog, if this guy is the culprit, then it will work. 
You may try also [this core](https://github.com/BrunoLevy/TordBoyau) (that I developed under Vivado), but I have not written the documentation for it yet.

usmank11, 2023-01-08:
Thanks for the support! I'll follow along with the tutorial and start testing from step 20. I'll also take a peek at your new core.

BrunoLevy, 2023-01-08:
I'm currently writing a small test program to try to figure out what happens with Vivado, will keep you updated.

BrunoLevy, 2023-01-08:
Pushed `step7_with_disasm.v`, to test whether my risc-V assembler in VERILOG works with VIVADO.
Here is what it displays under Icarus and Verilator. Would you tell me whether you get the same thing ?

```
PC=          0 nop
PC=          1 add x1,x0,x0
PC=          2 addi x1,x1,1
PC=          3 addi x1,x1,1
PC=          4 addi x1,x1,1
PC=          5 addi x1,x1,1
PC=          6 add x2,x1,x0
PC=          7 add x3,x1,x2
PC=          8 srli x3,x3,3
PC=          9 slli x3,x3,31
PC=         10 srai x3,x3,5
PC=         11 srli x1,x3,26
PC=         12 ebreak
ALUreg rd= 0 rs1= 0 rs2= 0 funct3=000
ALUreg rd= 1 rs1= 0 rs2= 0 funct3=000
x1 <= 00000000000000000000000000000000
ALUimm rd= 1 rs1= 1 imm=1 funct3=000
x1 <= 00000000000000000000000000000001
LEDS = 00001
ALUimm rd= 1 rs1= 1 imm=1 funct3=000
x1 <= 00000000000000000000000000000010
LEDS = 00010
ALUimm rd= 1 rs1= 1 imm=1 funct3=000
x1 <= 00000000000000000000000000000011
LEDS = 00011
ALUimm rd= 1 rs1= 1 imm=1 funct3=000
x1 <= 00000000000000000000000000000100
LEDS = 00100
ALUreg rd= 2 rs1= 1 rs2= 0 funct3=000
x2 <= 00000000000000000000000000000100
ALUreg rd= 3 rs1= 1 rs2= 2 funct3=000
x3 <= 00000000000000000000000000001000
ALUimm rd= 3 rs1= 3 imm=3 funct3=101
x3 <= 00000000000000000000000000000001
ALUimm rd= 3 rs1= 3 imm=31 funct3=001
x3 <= 10000000000000000000000000000000
ALUimm rd= 3 rs1= 3 imm=1029 funct3=101
x3 <= 11111100000000000000000000000000
ALUimm rd= 1 rs1= 3 imm=26 funct3=101
x1 <= 00000000000000000000000000111111
LEDS = 11111
SYSTEM
```

usmank11, 2023-01-08:
Getting this small error in the code.
[image]
Fixed it by changing to myPC = myPC + 1.

Simulation still shows the X's and nothing when i upload to board. Console shows this:
[image]

BrunoLevy, 2023-01-08:
Thank you for testing ! (pushed the fixes)

From the output: the assembler and disassembler both work, so we still don't know who is the culprit !
Where do you see the X's ? Is it in another window ?
Is it possible to plug a waveform display on LEDS[4:0] ? Maybe doing that, you will see whether there is activity there.

usmank11, 2023-01-08:
In the waveform I don't see any activity on the LED's.
[image]

BrunoLevy, 2023-01-08:
I pushed a new version of `step7_with_disasm.v`, that displays the current instruction being executed in BENCH mode.
Here is what it displays under Icarus and Verilator.
What does it give under Vivado ?
```
PC=         0 nop
ALUreg rd= 0 rs1= 0 rs2= 0 funct3=000
PC=         4 add x1,x0,x0
ALUreg rd= 1 rs1= 0 rs2= 0 funct3=000
x1 <= 00000000000000000000000000000000
PC=         8 addi x1,x1,1
ALUimm rd= 1 rs1= 1 imm=1 funct3=000
x1 <= 00000000000000000000000000000001
LEDS = 00001
PC=        12 addi x1,x1,1
ALUimm rd= 1 rs1= 1 imm=1 funct3=000
x1 <= 00000000000000000000000000000010
LEDS = 00010
PC=        16 addi x1,x1,1
ALUimm rd= 1 rs1= 1 imm=1 funct3=000
x1 <= 00000000000000000000000000000011
LEDS = 00011
PC=        20 addi x1,x1,1
ALUimm rd= 1 rs1= 1 imm=1 funct3=000
x1 <= 00000000000000000000000000000100
LEDS = 00100
PC=        24 add x2,x1,x0
ALUreg rd= 2 rs1= 1 rs2= 0 funct3=000
x2 <= 00000000000000000000000000000100
PC=        28 add x3,x1,x2
ALUreg rd= 3 rs1= 1 rs2= 2 funct3=000
x3 <= 00000000000000000000000000001000
PC=        32 srli x3,x3,3
ALUimm rd= 3 rs1= 3 imm=3 funct3=101
x3 <= 00000000000000000000000000000001
PC=        36 slli x3,x3,31
ALUimm rd= 3 rs1= 3 imm=31 funct3=001
x3 <= 10000000000000000000000000000000
PC=        40 srai x3,x3,5
ALUimm rd= 3 rs1= 3 imm=1029 funct3=101
x3 <= 11111100000000000000000000000000
PC=        44 srli x1,x3,26
ALUimm rd= 1 rs1= 3 imm=26 funct3=101
x1 <= 00000000000000000000000000111111
LEDS = 11111
PC=        48 ebreak
SYSTEM
```

usmank11, 2023-01-08:
No output on my end. I also had to change "++i" on line 98 to "i=i+1" so i don't have any errors.
[image]

BrunoLevy, 2023-01-09:
So I suspect simulation does not simulate the clock (which would explain why nothing happens). 
Was it working with steps 1-6 ?

usmank11, 2023-01-09:
The simulation does show the clock working. Steps 1-6 all worked fine.

danielgusvt, 2025-09-22:
I hope I am not necrobumping this issue, but I experienced the same problem with Quartus toolchain, I tried to narrow this issue down and found the same as the previous comment said, that memory can't be inferred if we use `task` inside `initial begin`.