BrunoLevy/learn-fpga #94: Which linker script should I be using with the ULX3S?

JamesTimothyMeech, 2023-04-21:
With the current version of the repo when I attempt to compile a C program and load it onto the ULX3S the Makefile points to a file that does not exist: 

`cannot open linker script file /home/james/Desktop/Casino/learn-fpga/FemtoRV/FIRMWARE//CRT/spiflash_ulx3s.ld: No such file or directory`

which linker script should I be using with the ULX3S?

BrunoLevy, 2023-04-21:
You need to configure the firmware for `ULX3S` before compiling your C program, as follows:
```
$ cd learn-fpga/FemtoRV
$ make ULX3S.firware_config 
```

This will create the file `FIRMWARE/config.mk` with compilation flags and defines.

Then there are two options to load software on the ULX3S, either use FemtOS, and create `.elf` binaries stored on the SDCard, more details in the [tutorial](https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/TUTORIALS/ULX3S.md), or generate `.hex` files directly incorporated in your design. You can use the Mafile from `FIRMWARE/FEMTOS` as a starting point for writing your own baremetal software.

JamesTimothyMeech, 2023-04-28:
Thanks when I wrote the issue it was not clear to me that I should be loading .elf binaries on an SD card to run them on the ULX3S.

closed 2023-04-28 by JamesTimothyMeech