BrunoLevy/learn-fpga #99: Struggling to make LiteOS

JamesTimothyMeech, 2023-05-10:
When I run `make` for LiteOS (after fixing some errors I got from unescaped "" in a printf statement) I get this error:

```ar cq libliteos.a lite_oled.o lite_fb.o lite_elf.o lite_stdio.o imgui.o imgui_demo.o imgui_draw.o imgui_tables.o imgui_widgets.o imgui_sw.o 
ranlib libliteos.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: archive library: libliteos.a the table of contents is empty (no object file members in the library define global symbols)
 CC       boot.elf
/usr/local/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: boot.elf has a LOAD segment with RWX permissions
/usr/local/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /Users/jamesmeech/Desktop/Casino/learn-fpga/LiteX/build/digilent_arty/software/liblitedram/liblitedram.a(sdram.o): in function `sdram_init':
/Users/jamesmeech/Desktop/Casino/Enjoy_Digital/litex/litex/soc/software/liblitedram/sdram.c:1179: undefined reference to `memtest'
/usr/local/Cellar/riscv-gnu-toolchain/main/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /Users/jamesmeech/Desktop/Casino/learn-fpga/LiteX/build/digilent_arty/software/liblitedram/liblitedram.a(sdram.o): in function `csr_write_simple':
/Users/jamesmeech/Desktop/Casino/Enjoy_Digital/litex/litex/soc/software/include/hw/common.h:33: undefined reference to `memspeed'
collect2: error: ld returned 1 exit status
make: *** [boot.elf] Error 1```

How can I fix this?

JamesTimothyMeech, 2023-06-03:
I'm very keen to hear any advice on how to proceed with debugging this as it is blocking my progress on an important project!

BrunoLevy, 2023-06-03:
I'll take a look asap. Seems that smthg was changed in LiteX that breaks LiteOS compilation.

JamesTimothyMeech, 2023-06-08:
Although I have to revert to the newest version of LiteX if I want commands like this: 
`python3 -m litex_boards.targets.gsd_orangecrab --cpu-type femtorv --cpu-variant petitbateau --build --load --device 25F --sdram-device MT41K64M16 --ecppack-compress --with-spi-sdcard`

BrunoLevy, 2023-06-08:
Yes, using an old version of LiteX is just a temporary fix, I'm trying to fix everything for the newest version (for now I have a couple of things to do with the version for the ULX3S that has a few add-ons that no longer work with the newest LiteX, in particular the ability to switch the SDCard between the ECP5 and the ESP32 dynamically, which is useful to send files to the SDCard through WIFI directly...)

JamesTimothyMeech, 2023-06-09:
Thanks for your help with this! So far I have got the LiteOs binary that I managed to compile running on an orangecrab 85-F by copying boot.bin to the SD card (I am waiting to get my hands on a ULX3S unfortunately) but it appears to hang before I get the LiteOs prompt: 
`jamesmeech@Jamess-MacBook-Pro-10 LiteX % litex_term /dev/tty.usbmodem143101

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun  8 2023 18:06:45
 BIOS CRC passed (f0753371)

 LiteX git sha1: c6adf703

--=============== SoC ==================--
CPU:		FemtoRV-PETITBATEAU @ 48MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		128.0MiB 16-bit @ 192MT/s (CL-6 CWL-5)
MAIN-RAM:	128.0MiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |01110000| delays: 02+-01
  m0, b01: |00000000| delays: -
  m0, b02: |00000000| delays: -
  m0, b03: |00000000| delays: -
  best: m0, b00 delays: 02+-01
  m1, b00: |01110000| delays: 02+-01
  m1, b01: |00000000| delays: -
  m1, b02: |00000000| delays: -
  m1, b03: |00000000| delays: -
  best: m1, b00 delays: 02+-01
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 7.8MiB/s
   Read speed: 11.7MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SPI-Mode...
Booting from boot.json...
boot.json file not found.
Booting from boot.bin...
Copying boot.bin to 0x40000000 (14864 bytes)...
[########################################]
Executing booted program at 0x40000000

--============= Liftoff! ===============--`
The output unfortunately stops there.

BrunoLevy, 2023-06-09:
Different things you can try:
- try with a simpler CPU variant (electron or gracilis). Do not forget to recompile boot.bin and copy it to the SDCard 
- try to run the [demobundle](https://github.com/BrunoLevy/learn-fpga/tree/master/LiteX/software/DemoBundle)

BrunoLevy, 2023-06-09:
(I have a 45F orange crab at home, I'll try with it as soon as I have time)

JamesTimothyMeech, 2023-06-09:
I tried the electron and got the same issue: 
```
jamesmeech@Jamess-MacBook-Pro-10 LiteOS % litex_term /dev/tty.usbmodem143101

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun  9 2023 13:26:09
 BIOS CRC passed (ca08dc12)

 LiteX git sha1: c6adf703

--=============== SoC ==================--
CPU:		FemtoRV-ELECTRON @ 48MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		128.0MiB 16-bit @ 192MT/s (CL-6 CWL-5)
MAIN-RAM:	128.0MiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |01110000| delays: 02+-01
  m0, b01: |00000000| delays: -
  m0, b02: |00000000| delays: -
  m0, b03: |00000000| delays: -
  best: m0, b00 delays: 02+-01
  m1, b00: |01110000| delays: 02+-01
  m1, b01: |00000000| delays: -
  m1, b02: |00000000| delays: -
  m1, b03: |00000000| delays: -
  best: m1, b00 delays: 02+-01
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 7.8MiB/s
   Read speed: 13.5MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SPI-Mode...
Booting from boot.json...
boot.json file not found.
Booting from boot.bin...
Copying boot.bin to 0x40000000 (14864 bytes)...
[########################################]
Executing booted program at 0x40000000

--============= Liftoff! ===============--
```

I'll try the demobudle on verriscv.

JamesTimothyMeech, 2023-06-09:
I tried with electron after recompiling boot.bin and copying it the the SDcard but I had the same problem: 
```
jamesmeech@Jamess-MacBook-Pro-10 LiteOS % litex_term /dev/tty.usbmodem143101

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun  9 2023 13:26:09
 BIOS CRC passed (ca08dc12)

 LiteX git sha1: c6adf703

--=============== SoC ==================--
CPU:		FemtoRV-ELECTRON @ 48MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		128.0MiB 16-bit @ 192MT/s (CL-6 CWL-5)
MAIN-RAM:	128.0MiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |01110000| delays: 02+-01
  m0, b01: |00000000| delays: -
  m0, b02: |00000000| delays: -
  m0, b03: |00000000| delays: -
  best: m0, b00 delays: 02+-01
  m1, b00: |01110000| delays: 02+-01
  m1, b01: |00000000| delays: -
  m1, b02: |00000000| delays: -
  m1, b03: |00000000| delays: -
  best: m1, b00 delays: 02+-01
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 7.8MiB/s
   Read speed: 13.5MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SPI-Mode...
Booting from boot.json...
boot.json file not found.
Booting from boot.bin...
Copying boot.bin to 0x40000000 (14816 bytes)...
[########################################]
Executing booted program at 0x40000000

--============= Liftoff! ===============--
```

I'll try vexriscv with the demobudle next.

JamesTimothyMeech, 2023-06-09:
I'm having trouble compiling demobundle for vexriscv_smp so I'll try FemtoRV again: 
```
jamesmeech@Jamess-MacBook-Pro-10 DemoBundle % make
 CC       tinyraytracer.o
 CC       mandelbrot.o
 CC       oled_test.o
demos/oled_test.c:8:13: warning: 'oled_test' defined but not used [-Wunused-function]
    8 | static void oled_test(int nb_args, char** args) {
      |             ^~~~~~~~~
 CC       oled_riscv_logo.o
demos/oled_riscv_logo.c:69:13: warning: 'oled_riscv_logo' defined but not used [-Wunused-function]
   69 | static void oled_riscv_logo(int nb_args, char** args) {
      |             ^~~~~~~~~~~~~~~
 CC       oled_julia.o
demos/oled_julia.c:49:13: warning: 'oled_julia' defined but not used [-Wunused-function]
   49 | static void oled_julia(int nb_args, char** args) {
      |             ^~~~~~~~~~
 CC       pi.o
 CC       complete.o
 CC       helpers.o
 CC       readline.o
 CC       cmd_bios.o
 CC       commands.o
 CC       isr.o
 CC       main.o
 CC       crt0.o
 CC       lite_oled.o
../Libs/lite_oled.c: In function 'oled_init':
../Libs/lite_oled.c:10:8: warning: unused variable 'trigger_warning' [-Wunused-variable]
   10 |    int trigger_warning; // LiteX synthesized without OLED support (graphics commands will be ignored)
      |        ^~~~~~~~~~~~~~~
 CC       lite_fb.o
 CC       lite_elf.o
 CC       lite_stdio.o
../Libs/lite_stdio.c: In function 'lx_mount':
../Libs/lite_stdio.c:82:9: warning: implicit declaration of function 'fatfs_set_ops_spisdcard' [-Wimplicit-function-declaration]
   82 |         fatfs_set_ops_spisdcard();
      |         ^~~~~~~~~~~~~~~~~~~~~~~
 CX       imgui.o
 CX       imgui_demo.o
 CX       imgui_draw.o
 CX       imgui_tables.o
 CX       imgui_widgets.o
 CX       imgui_sw.o
ar cq libliteos.a lite_oled.o lite_fb.o lite_elf.o lite_stdio.o imgui.o imgui_demo.o imgui_draw.o imgui_tables.o imgui_widgets.o imgui_sw.o 
ranlib libliteos.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: archive library: libliteos.a the table of contents is empty (no object file members in the library define global symbols)
 CC       demo.elf
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: warning: demo.elf has a LOAD segment with RWX permissions
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccwqIwV1.ltrans0.ltrans.o: in function `.L39':
<artificial>:(.text+0x2ea): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccwqIwV1.ltrans0.ltrans.o: in function `.L51':
<artificial>:(.text+0x3d4): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0x406): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0x448): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccwqIwV1.ltrans0.ltrans.o: in function `.L99':
<artificial>:(.text+0x548): undefined reference to `oled_init'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccwqIwV1.ltrans0.ltrans.o: in function `leds_handler':
<artificial>:(.text+0xe50): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0xe78): undefined reference to `__truncdfsf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccwqIwV1.ltrans0.ltrans.o: in function `.L220':
<artificial>:(.text+0xfe4): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0x1004): undefined reference to `__truncdfsf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccwqIwV1.ltrans0.ltrans.o: in function `.L219':
<artificial>:(.text+0x1052): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0x1072): undefined reference to `__truncdfsf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccwqIwV1.ltrans0.ltrans.o: in function `.L245':
<artificial>:(.text+0x140a): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccwqIwV1.ltrans0.ltrans.o: in function `.L260':
<artificial>:(.text+0x1432): undefined reference to `__truncdfsf2'
[... 71 lines cut ...]
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccwqIwV1.ltrans0.ltrans.o: in function `.L220':
<artificial>:(.text+0xfe4): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0x1004): undefined reference to `__truncdfsf2'
[... cut ...]
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /Users/jamesmeech/Desktop/Casino/learn-fpga/LiteX/build/radiona_ulx3s/software/libc/libc.a(libm_math_sf_pow.c.o): in function `powf':
/Users/jamesmeech/Desktop/Casino/learn-fpga/LiteX/build/radiona_ulx3s/software/libc/../../../../../../Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libm/math/sf_pow.c:165: undefined reference to `__truncdfsf2'
collect2: error: ld returned 1 exit status
make: *** [demo.elf] Error 1
```

BrunoLevy, 2023-06-09:
Are you using LiteX version [f8863bb](https://github.com/enjoy-digital/litex/commit/f8863bb986fc448baa1b086f1bafea52e4685c91) ?

JamesTimothyMeech, 2023-06-09:
It seems to produce the same error: 
```
jamesmeech@Jamess-MacBook-Pro-10 DemoBundle % make clean
rm -f *.d *.o *.a *.elf *.list .*~ *~
jamesmeech@Jamess-MacBook-Pro-10 DemoBundle % make      
 CC       tinyraytracer.o
 CC       mandelbrot.o
 CC       oled_test.o
demos/oled_test.c:8:13: warning: 'oled_test' defined but not used [-Wunused-function]
    8 | static void oled_test(int nb_args, char** args) {
      |             ^~~~~~~~~
 CC       oled_riscv_logo.o
demos/oled_riscv_logo.c:69:13: warning: 'oled_riscv_logo' defined but not used [-Wunused-function]
   69 | static void oled_riscv_logo(int nb_args, char** args) {
      |             ^~~~~~~~~~~~~~~
 CC       oled_julia.o
demos/oled_julia.c:49:13: warning: 'oled_julia' defined but not used [-Wunused-function]
   49 | static void oled_julia(int nb_args, char** args) {
      |             ^~~~~~~~~~
 CC       pi.o
 CC       complete.o
 CC       helpers.o
 CC       readline.o
 CC       cmd_bios.o
 CC       commands.o
 CC       isr.o
 CC       main.o
 CC       crt0.o
 CC       lite_oled.o
../Libs/lite_oled.c: In function 'oled_init':
../Libs/lite_oled.c:10:8: warning: unused variable 'trigger_warning' [-Wunused-variable]
   10 |    int trigger_warning; // LiteX synthesized without OLED support (graphics commands will be ignored)
      |        ^~~~~~~~~~~~~~~
 CC       lite_fb.o
 CC       lite_elf.o
 CC       lite_stdio.o
../Libs/lite_stdio.c: In function 'lx_mount':
../Libs/lite_stdio.c:82:9: warning: implicit declaration of function 'fatfs_set_ops_spisdcard' [-Wimplicit-function-declaration]
   82 |         fatfs_set_ops_spisdcard();
      |         ^~~~~~~~~~~~~~~~~~~~~~~
 CX       imgui.o
 CX       imgui_demo.o
 CX       imgui_draw.o
 CX       imgui_tables.o
 CX       imgui_widgets.o
 CX       imgui_sw.o
ar cq libliteos.a lite_oled.o lite_fb.o lite_elf.o lite_stdio.o imgui.o imgui_demo.o imgui_draw.o imgui_tables.o imgui_widgets.o imgui_sw.o 
ranlib libliteos.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: archive library: libliteos.a the table of contents is empty (no object file members in the library define global symbols)
 CC       demo.elf
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: warning: demo.elf has a LOAD segment with RWX permissions
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccVJA33P.ltrans0.ltrans.o: in function `.L39':
<artificial>:(.text+0x2ea): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccVJA33P.ltrans0.ltrans.o: in function `.L51':
<artificial>:(.text+0x3d4): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0x406): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0x448): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccVJA33P.ltrans0.ltrans.o: in function `.L99':
<artificial>:(.text+0x548): undefined reference to `oled_init'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccVJA33P.ltrans0.ltrans.o: in function `leds_handler':
<artificial>:(.text+0xe50): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0xe78): undefined reference to `__truncdfsf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccVJA33P.ltrans0.ltrans.o: in function `.L220':
<artificial>:(.text+0xfe4): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0x1004): undefined reference to `__truncdfsf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccVJA33P.ltrans0.ltrans.o: in function `.L219':
<artificial>:(.text+0x1052): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: <artificial>:(.text+0x1072): undefined reference to `__truncdfsf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccVJA33P.ltrans0.ltrans.o: in function `.L245':
<artificial>:(.text+0x140a): undefined reference to `__extendsfdf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccVJA33P.ltrans0.ltrans.o: in function `.L260':
<artificial>:(.text+0x1432): undefined reference to `__truncdfsf2'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /var/folders/p7/23012dkx5dd9vcjx2glbl9mh0000gn/T//ccVJA33P.ltrans0.ltrans.o: in function `run_tinyraytracer':
<artificial>:(.text+0x1bb6): undefined reference to `oled_init'
[... cut ...]
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /Users/jamesmeech/Desktop/Casino/learn-fpga/LiteX/build/radiona_ulx3s/software/libc/libc.a(libm_math_sf_pow.c.o): in function `powf':
/Users/jamesmeech/Desktop/Casino/learn-fpga/LiteX/build/radiona_ulx3s/software/libc/../../../../../../Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libm/math/sf_pow.c:165: undefined reference to `__truncdfsf2'
collect2: error: ld returned 1 exit status
make: *** [demo.elf] Error 1
```

JamesTimothyMeech, 2023-06-09:
It seems to be happy to run the litex demo.bin file: 
```
jamesmeech@Jamess-MacBook-Pro-10 demo % litex_term /dev/tty.usbmodem143101

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun  9 2023 14:04:22
 BIOS CRC passed (c05077c0)

 LiteX git sha1: c6adf703

--=============== SoC ==================--
CPU:		FemtoRV-GRACILIS @ 48MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		128.0MiB 16-bit @ 192MT/s (CL-6 CWL-5)
MAIN-RAM:	128.0MiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |01110000| delays: 02+-01
  m0, b01: |00000000| delays: -
  m0, b02: |00000000| delays: -
  m0, b03: |00000000| delays: -
  best: m0, b00 delays: 02+-01
  m1, b00: |01110000| delays: 02+-01
  m1, b01: |00000000| delays: -
  m1, b02: |00000000| delays: -
  m1, b03: |00000000| delays: -
  best: m1, b00 delays: 02+-01
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 8.7MiB/s
   Read speed: 13.6MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SPI-Mode...
Booting from boot.json...
boot.json file not found.
Booting from boot.bin...
boot.bin file not found.
SDCard boot failed.
No boot medium found

--============= Console ================--

litex> help

LiteX BIOS, available commands:

leds                     - Set Leds value
flush_l2_cache           - Flush L2 cache
flush_cpu_dcache         - Flush CPU data cache
crc                      - Compute CRC32 of a part of the address space
ident                    - Identifier of the system
help                     - Print this help

sdcardboot               - Boot from SDCard
serialboot               - Boot from Serial (SFL)
reboot                   - Reboot
boot                     - Boot from Memory

mem_cmp                  - Compare memory content
mem_speed                - Test memory speed
mem_test                 - Test memory access
mem_copy                 - Copy address space
mem_write                - Write address space
mem_read                 - Read address space
mem_list                 - List available memory regions

sdram_mr_write           - Write SDRAM Mode Register
sdram_cal                - Calibrate SDRAM
sdram_test               - Test SDRAM
sdram_init               - Initialize SDRAM (Init + Calibration)
```

BrunoLevy, 2023-06-09:
The messages that you see come from the BIOS, that is bundled with the bitstream sent to the FPGA.
It says "boot.bin: file not found", it seems that for some reason the SDCard does not work.
You can try running `demo .bin` by sending it through `litex_term`:
```
$ litex_term --kernel demo.bin /dev/ttyUSBnn
litex> reboot
```
When you invoke reboot, it restarts the code that downloads `demo.bin` through the serial connection then starts it.

JamesTimothyMeech, 2023-06-12:
Is there any way to debug my SD card setup? It is a brand new SanDisk SD card that is FAT32 formatted and has nothing but the binary on there?

JamesTimothyMeech, 2023-06-12:
Should mem_list show the SD card? 
```

litex> mem_list  
Available memory regions:
ROM       0x00000000 0x20000 
SRAM      0x01000000 0x2000 
MAIN_RAM  0x40000000 0x8000000 
CSR       0x82000000 0x10000 
```

BrunoLevy, 2023-06-13:
Hi, 
I had to struggle too a while ago, and I think I remember two things:
- (1) make sure you synthesize with SPI SDCard enabled (you probably checked already, I'm saying that just in case...)
- (2) there are two different drivers for the sdcard, one of top of a generic spi driver (spi_sdcard) and another one (sdcard). In LiteOS I'm using spi_sdcard

BrunoLevy, 2023-06-13:
Do not give up ! I have a little bit of time this week, and I understood compilation problems with latest LiteX,
if things go well, at least I'll have everything working on my ULX3S this week, will keep you updated.

BrunoLevy, 2023-06-13:
To compile LiteOS with latest LiteX:
edit `build/<platform>/software/include/generated/variables.mak`
and replace in `CPU_FLAGS` the string `rv32i2p0_mc` with `rv32i` or `rv32im` (depending on CPU)
Then it will link
It will still complain about a missing `memtest` symbol, I'm investigating...

BrunoLevy, 2023-06-13:
Pushed a new `learn-fpga/LiteX/software/makefile.inc` that works with up-to-date LiteX
- links with the right `libgcc` (no longer need to manually edit `generated/variables.mak` like in previous post)
- links with `libbase`, that has now `memspeed()` and `memtest()` 
I'll test that tomorrow with my ULX3S (and also OrangeCrab as soon as I have time). We are nearly there...

BrunoLevy, 2023-06-13:
On my ULX3S, with the new `makefile.inc` I could successfully synthesize a `femtorv-gracilis`, compile `LiteOS` for it, boot it from the SDCard and start example programs. 
I'll try with an OrangeCrab tomorrow.

BrunoLevy, 2023-06-13:
Note: C++ programs do not link (something remains to be fixed in `makefile.inc`, to be investigated)

BrunoLevy, 2023-06-14:
On the ULX3S, a small modification in one of LiteX files needs to be done to let the FPGA access the SDCard, I have filed a [LiteX issue](https://github.com/enjoy-digital/litex/issues/1712)  (all details given there if you want to try it)

BrunoLevy, 2023-06-14:
I confirm that LiteOS works on my OrangeCrab.

Synthesize and program:
```
$ cd learn-fpga/LiteX
$ python3 -m litex_boards.targets.gsd_orangecrab --device 25F --cpu-type femtorv --cpu-variant gracilis --with-spi-sdcard --build --load --ecppack-compress
```

Build software:
Edit `software/makefile.inc`
- set `LITEX_DIR` and `LEARNFPGA_DIR`
- set  `LITEX_PLATFORM` to `gsd_orangecrab` 
```
$ cd software/LiteOS
$ make
$ cd ../Programs
$ make hello.elf
```

Copy `LiteOS/boot.bin` and `Programs/hello.elf` to a SDCard
Insert the SDCard in the OrangeCrab

```
$ litex_term /dev/ttyACM0
$ reboot
```

This should load LiteOS from the SDCard

```
liteOS> run hello.elf
```

JamesTimothyMeech, 2023-06-14:
Awesome! It also works on my 85F Orangecrab! Thanks so much for your help with this! I will close the issue now!

closed 2023-06-14 by JamesTimothyMeech