BrunoLevy/learn-fpga #111: Rv32i quark Block diagram

cicamargoba, 2023-07-31:
[rv32i_block_diagram (1).pdf](https://github.com/BrunoLevy/learn-fpga/files/12210764/rv32i_block_diagram.1.pdf)
This block diagram is Right for you?

Carlos

Mecrisp, 2023-07-31:
On WMASK multiplexer, the Half1 and Half2 should be 0011 and 1100. In the diagram, both are 1100.
Also I think the input to loadstore_addr should be [31:0] instead of just [23:0]. True, one can synthesise the core with smaller address width; but you are using [31:0] as input for new PC in the diagram, which does not mix. The new PC multiplexer has three input lines in the diagram, but only two are annotated.

So far my first observations, there might be more I just haven't spotted on first try.

cicamargoba, 2023-07-31:
Hello, Thanks for your feedback.
I've already fix the diagram based on your review
[rv32i_block_diagram.pdf](https://github.com/BrunoLevy/learn-fpga/files/12215713/rv32i_block_diagram.pdf)
iew. I'm attaching the SoC block diagram.
I'm teacher and I'll use you
[soc_femto_block_diagram.pdf](https://github.com/BrunoLevy/learn-fpga/files/12215716/soc_femto_block_diagram.pdf)
r excellent RV32I description.

cicamargoba, 2023-07-31:
[output.pdf](https://github.com/BrunoLevy/learn-fpga/files/12215925/output.pdf)
Thanks for your comments
You're right, show the whole architecture will be a bad idea for students.
I'm attaching some diagrams explaining how the architecture deal with different instructions, and I add some simulations too.
Very good Idea, use different colors for diagrams.