BrunoLevy/learn-fpga #112: Out of memory while generating hex for blinky in step 20, IceStick

dbenn, 2023-08-05:
I see this error when attempting to generate the hex file for `blinky.bram.elf`:
~~~
firmware_words blinker.bram.elf -ram 6144 -max_addr 6144 -out firmware.hex
   RAM SIZE=6144
   LOAD ELF: blinker.bram.elf
       max address=65676
   RAM_SIZE: 6144
Memory exceeded !
~~~
Same thing if `max_addr` is 65676. Thoughts? I can show how  I build `firmware_words`, if it helps.

Would it instead be possible to do this?
~~~
riscv32-unknown-elf-objcopy --change-address 0x00000000 -O verilog blinker.bram.elf blinker.bram.hex
~~~
This generates the hex file, which In can use as `firmware.hex` for step 20. It loads but the run (on simulator and IceStick) yields no LEDs output.

BrunoLevy, 2023-08-05:
Hi,
Did you first try using the scripts and steps described in the tutorial ?
(my script exist for many different boards and configurations, so if you do not strictly follow the tutorial, you may generate an executable for another configuration). In particular, this concerns the linker script used to generate the ELF.

dbenn, 2023-08-05:
Hi @BrunoLevy and apologies. I've just resolved the problem. I was initially trying a docker container with the `riscv` toolchain because I thought it was not present or incorrectly installed on my Mac.

After using `make blinker.bram.hex` rather than the individual build commands, it now works fine, correctly detecting the toolchain architecture with the code running via `run.sh`, `verilator` and on the IceStick.

Thanks for your reply and your excellent tutorial. I've successfully followed all previous steps, incorporating a Pmod dual 7-segment LED display in a number of the steps, including as a device in step 17.

BrunoLevy, 2023-08-05:
Thank you very much for letting me know,
and congratulation for extending it to a new device !
Cheers,
-- Bruno

closed 2023-08-05 by BrunoLevy