BrunoLevy/learn-fpga #125: Make ICESTICK fail with latest oss-cad-suite for macOS 12

daveti, 2024-12-27:
⦗OSS CAD Suite⦘ Mac-Pro:FemtoRV daveti$ make ICESTICK
BOARD=icestick TOOLS/make_config.sh -DICE_STICK
./PROCESSOR/femtorv32_quark.v:63: error: Unable to bind wire/reg/memory `instr['sd11:'sd7]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:63: error: Unable to elaborate r-value: instr['sd11:'sd7]
./PROCESSOR/femtorv32_quark.v:68: error: Unable to bind wire/reg/memory `instr['sd14:'sd12]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:68: error: Unable to elaborate r-value: (8'd1)<<(instr['sd14:'sd12])
./PROCESSOR/femtorv32_quark.v:71: error: Unable to bind wire/reg/memory `instr['sd31]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:71: error: Unable to bind wire/reg/memory `instr['sd30:'sd12]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:72: error: Unable to bind wire/reg/memory `instr['sd31]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:72: error: Concatenation/replication may not have zero width in this context.
./PROCESSOR/femtorv32_quark.v:72: error: Unable to bind wire/reg/memory `instr['sd30:'sd20]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:72: error: Concatenation/replication may not have zero width in this context.
./PROCESSOR/femtorv32_quark.v:72: error: Unable to elaborate r-value: {{'sd21{instr['sd31]}}, instr['sd30:'sd20]}
./PROCESSOR/femtorv32_quark.v:74: error: Unable to bind wire/reg/memory `instr['sd31]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:74: error: Concatenation/replication may not have zero width in this context.
./PROCESSOR/femtorv32_quark.v:74: error: Unable to bind wire/reg/memory `instr['sd30:'sd25]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:74: error: Unable to bind wire/reg/memory `instr['sd11:'sd7]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:74: error: Concatenation/replication may not have zero width in this context.
./PROCESSOR/femtorv32_quark.v:74: error: Unable to elaborate r-value: {{'sd21{instr['sd31]}}, instr['sd30:'sd25], instr['sd11:'sd7]}
./PROCESSOR/femtorv32_quark.v:75: error: Unable to bind wire/reg/memory `instr['sd31]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:75: error: Concatenation/replication may not have zero width in this context.
./PROCESSOR/femtorv32_quark.v:75: error: Unable to bind wire/reg/memory `instr['sd7]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:75: error: Unable to bind wire/reg/memory `instr['sd30:'sd25]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:75: error: Unable to bind wire/reg/memory `instr['sd11:'sd8]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:76: error: Unable to bind wire/reg/memory `instr['sd31]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:76: error: Concatenation/replication may not have zero width in this context.
./PROCESSOR/femtorv32_quark.v:76: error: Unable to bind wire/reg/memory `instr['sd19:'sd12]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:76: error: Unable to bind wire/reg/memory `instr['sd20]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:76: error: Unable to bind wire/reg/memory `instr['sd30:'sd21]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:80: error: Unable to bind wire/reg/memory `instr['sd6:'sd2]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:80: error: Unable to elaborate r-value: (instr['sd6:'sd2])==(5'd0)
./PROCESSOR/femtorv32_quark.v:81: error: Unable to bind wire/reg/memory `instr['sd6:'sd2]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:81: error: Unable to elaborate r-value: (instr['sd6:'sd2])==(5'd4)
./PROCESSOR/femtorv32_quark.v:82: error: Unable to bind wire/reg/memory `instr['sd6:'sd2]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:82: error: Unable to elaborate r-value: (instr['sd6:'sd2])==(5'd8)
./PROCESSOR/femtorv32_quark.v:83: error: Unable to bind wire/reg/memory `instr['sd6:'sd2]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:83: error: Unable to elaborate r-value: (instr['sd6:'sd2])==(5'd12)
./PROCESSOR/femtorv32_quark.v:84: error: Unable to bind wire/reg/memory `instr['sd6:'sd2]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:84: error: Unable to elaborate r-value: (instr['sd6:'sd2])==(5'd28)
./PROCESSOR/femtorv32_quark.v:85: error: Unable to bind wire/reg/memory `instr['sd3]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:85: error: Unable to elaborate r-value: instr['sd3]
./PROCESSOR/femtorv32_quark.v:86: error: Unable to bind wire/reg/memory `instr['sd6:'sd2]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:86: error: Unable to elaborate r-value: (instr['sd6:'sd2])==(5'd25)
./PROCESSOR/femtorv32_quark.v:87: error: Unable to bind wire/reg/memory `instr['sd6:'sd2]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:87: error: Unable to elaborate r-value: (instr['sd6:'sd2])==(5'd13)
[... 27 lines cut ...]
./PROCESSOR/femtorv32_quark.v:264:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:258: error: Unable to bind wire/reg/memory `LOAD_halfword' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:261:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:104: error: Unable to bind wire/reg/memory `writeBack' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:314:      : A symbol with that name was declared here. Check for declaration after use.
./PROCESSOR/femtorv32_quark.v:104: error: Unable to elaborate condition expression.
./PROCESSOR/femtorv32_quark.v:166: error: Unable to bind wire/reg/memory `instr['sd30]' in `FemtoRV32'
./PROCESSOR/femtorv32_quark.v:198:      : A symbol with that name was declared here. Check for declaration after use.
[... cut ...]
/Volumes/git2/learn-fpga/FemtoRV/FIRMWARE//TOOLCHAIN/xpack-riscv-none-embed-gcc-10.1.0-1.1-darwin-x64/bin/riscv-none-embed-as -march= -mabi=  -I/Volumes/git2/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOGL -I/Volumes/git2/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTORV32 -I/Volumes/git2/learn-fpga/FemtoRV/FIRMWARE//LIBFEMTOC  font_8x16.S -o font_8x16.o
Assembler messages:
Error: unrecognized option -%
make[2]: *** [font_8x16.o] Error 1
make[1]: *** [libs] Error 2
make: *** [ICESTICK.firmware_config] Error 2
⦗OSS CAD Suite⦘ Mac-Pro:FemtoRV daveti$

miguemoya, 2025-02-03:
Hello, this error also occurs on linux and in more files in the tutorial TUTORIALS/FROM_BLINKER_TO_RISCV/

You have to define wires and reg before you use.

You can see the changes at https://github.com/BrunoLevy/learn-fpga/compare/master...miguemoya:learn-fpga:master

But I can't get it to work when launching ./run_verilator.sh step18.v, the error messages are:

```
g++    ../../../FIRMWARE/LIBFEMTORV32/femto_elf.o sim_main.o verilated.o verilated_threads.o VSOC__ALL.a    -pthread -lpthread -latomic   -o VSOC
/usr/bin/ld: ../../../FIRMWARE/LIBFEMTORV32/femto_elf.o: relocations in generic ELF (EM: 243)
/usr/bin/ld: ../../../FIRMWARE/LIBFEMTORV32/femto_elf.o: error al añadir símbolos: file in wrong format
collect2: error: ld returned 1 exit status
make: *** [VSOC.mk:72: VSOC] Error 1
rm VSOC__ALL.verilator_deplist.tmp
```

PythonLinks, 2025-02-27:
I had the same problem, so I checked out the issues, and lo and behold, this is the most recent issue. 

So I clicked into that pull request, and found the original source for that pull request. I made a git branch. And tried it out. 

git pull https://github.com/miguemoya/learn-fpga

make ICESTICK worked just fine. 

Info: Device utilisation:
Info: 	         ICESTORM_LC:    1297/   1280   101%
Info: 	        ICESTORM_RAM:      16/     16   100%
Info: 	               SB_IO:      24/    112    21%
Info: 	               SB_GB:       7/      8    87%
Info: 	        ICESTORM_PLL:       1/      1   100%
Info: 	         SB_WARMBOOT:       0/      1     0%

The folloiwng 3 did not work. 

make ICEBREAKER

make ICESUGAR

make ICEFEATHER 

All broken.  I did not test all of the other options. 
I will wait a month, and try again. 

My goal is to fire up the bicycle version, and see how many resources it consumes, versus on Gatemate which has 4 input muxes.   Or compared to femtorv32_quark.v, which stores the registers in RAM instead of using lots of 4-Luts as multiplexers.

BrunoLevy, 2025-02-27:
I'll take a look asap, there are several things (mostly two things if I understand well):
- we use wires before declaring them, which is not accepted by all synthetizers, but this one is easy to fix
- there is a subtle issue about the way Yosys handles values read from BRAM, if I remember well the default behavior has changed and it now generates additional logic in case the value is used in the same cycle, which is not the case in FemtoRV. It requires some flags. I think @sylefeb also had the problem in Silice.

Gerard-170, 2025-05-08:
Hi all, 

I had the same problems using the main branch but all cleared out after declaring wires and nets before using. Working fine after that.