BrunoLevy/learn-fpga #129: Zero register x0 not initialized anywhere

MrJake222, 2025-05-12:
Hi,

The zero register (x0) is nowhere to be initialized. In `quark_bicycle` it's only set in a `initial` block. What about synthesis? Writing to it is also not protected in any way?

Mecrisp, 2025-05-12:
Hi! In real Lattice FPGAs, all the regs start with an initial value of zero, and the zero register is never written to. Explicitely setting x0 to zero in a reset clause is not necessary, but would introduce more LUTs into the design, and we are extremely tight on resources in the HX1K target which we wrote FemtoRV32-Quark for.
For the alushamt value, it may be on a random value from the run before the reset pulse, causing a few dummy shift cycles before the state machine fetches the first instruction. This does not harm further operation. We considered this small delay acceptable in exchange of more LUTs available for peripherals. We are aware of the issue in simulation, for which we offer the "BENCH" mode, as you already spotted.

MrJake222, 2025-05-12:
Hi, thanks for the response. I'm testing it on another Lattice FPGA namely iCE40UP5K. I'm not sure whether these regs end up equal to 0. As a compromise setting each one `reg whatever = 0;` is a viable option which I've been using and I get consistent simulation and synthesis behavior.

cicamargoba, 2026-08-16:
I have already added some initialization steps for the VLSI design; in this case, the registers start with random values ​​upon startup.

Carlos